RPi-Monitor/RPi-Monitor #334: Uninstalled rpi-monitor and now 'sudo apt update' no longer works.

bassamanator, 2020-07-19:
I uninstalled via sudo apt remove. Now when I try to update I get an error:
```
$ sudo apt update
Hit:1 http://raspbian.mirror.colo-serv.net/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Update rpimonitor Packages Status                         
sh: 1: /usr/share/rpimonitor/scripts/updatePackagesStatus.pl: not found
Reading package lists... Done
E: Problem executing scripts APT::Update::Post-Invoke 'echo 'Update rpimonitor Packages Status' && /usr/share/rpimonitor/scripts/updatePackagesStatus.pl'
E: Sub-process returned an error code
```

I manually removed /usr/share/rpimonitor _after_ I uninstalled rpimonitor via apt remove. This was an attempt to fix the apt update issue I was having.

How do I get back to normal here? Thanks.

gilarelli, 2020-07-22:
As you've uninstalled RPIMonitor, you can safely remove the file `/etc/apt/apt.conf.d/99rpimonitor` which is run after each apt update.

bassamanator, 2020-07-22:
@gilarelli Thanks!

closed 2020-07-22 by bassamanator