RPi-Monitor/RPi-Monitor #336: Updating fails on RPI with Arch ARM

TheZadok42, 2020-07-31:
When I run:
```bash
sudo /etc/init.d/rpimonitor update
```
I get the following output:
```bash
/etc/init.d/rpimonitor: line 14: /lib/lsb/init-functions: No such file or directory
```

Here is my system information:
```bash
➜  ~ sudo uname -a
Linux alarm 5.7.8-1-ARCH #1 SMP Sun Jul 12 03:38:28 UTC 2020 aarch64 GNU/Linux
```

I didn't found anything online, did I miss something super obvious?

Leepic, 2020-08-02:
Archlinux doesn't provide this file, it seem that ArchLinux is not LSB compliant. lsb-release only provide _lsb_release_ command, used to know your distribution release. Example:

```
% lsb_release -a
LSB Version:	1.4
Distributor ID:	Arch
Description:	Arch Linux
Release:	rolling
Codename:	n/a
```

Workaround:

- Comment the sourcing of LSB init-functions:

`nano +14 /etc/init.d/rpimonitor`

```bash
#. /lib/lsb/init-functions
```

Then, create this two functions after the previous commented line:

```bash
log_daemon_msg() {
        local msg
        msg="$1"
        echo "$msg"
}

log_end_msg() {
        declare -i retval
        retval="$1"
        return "$retval"
}
```

Finaly, inside the `update()` function, comment `apt-get` and add `pacman` command:

```bash
#  apt-get update >/dev/null 2>&1
pacman -S
```

Save and run again `/etc/init.d/rpimonitor update`.
After few seconds, you will see pending upgrades packages (if available) in your RPi-Monitor webUI.

TheZadok42, 2020-08-06:
This is amazing!  
Thank you very much!

TheZadok42, 2020-08-06:
I am now getting
```bash
Error: neither checkupdates nor aptitude seem to be available
```

DaniW42, 2020-08-12:
There is another file called `/usr/share/rpimonitor/scripts/updatePackagesStatus.pl` inside update(). This must be updated to fit your package manager. Maybe someone with Arch can help.

TheZadok42, 2020-08-30:
Thank you!  
This worked perfectly

closed 2020-08-30 by TheZadok42