RPi-Monitor/RPi-Monitor #341: web status dont refresh

frezeen, 2020-08-26:
when i open rpimonitor page its loook static, not autorefresh it every 10 secs , uptime clock seconds are static

from console this is the only alarm or error i have:
[Deprecation] Synchronous XMLHttpRequest on the main thread is deprecated because of its detrimental effects to the end user's experience. For more help, check https://xhr.spec.whatwg.org/.
send @ jquery.min.js:4

Leepic, 2020-11-29:
Have you checked "Auto refresh status page" in the "Options" menu?

gottaloveit, 2021-04-27:
There's an issue with the code calling synchronous ajax calls and jQuery deprecating them.  I got the "Options" menu to work finally on the status page by:

in file: 
`/usr/share/rpimonitor/web/js/rpimonitor.js`

at the end of the file, theres a function call:
`getVersion();`

simply comment it out like so:
`//getVersion();`

Save the file, refresh the page in browser and the menu in Options should have that option now.