RPi-Monitor/RPi-Monitor #342: How to add addon in docker container?

skywinder, 2020-08-28:
1. Started via: 
`docker run --device=/dev/vchiq --device=/dev/vcsm --volume=/opt/vc:/opt/vc --volume=/boot:/boot --volume=/sys:/dockerhost/sys:ro --volume=/etc:/dockerhost/etc:ro --volume=/proc:/dockerhost/proc:ro --volume=/usr/lib:/dockerhost/usr/lib:ro -p=8888:8888 --name="rpi-monitor" -d  michaelmiklis/rpi-monitor:latest`

I attached to docker via:

`docker exec -it XXX /bin/bash`

[image]

then I found that there is no text editor inside. So how and where should I edit config file?

smholvoet, 2020-08-30:
The config file should be located at `/etc/rpimonitor/data.conf` according do the 📖 [docs]( https://xavierberger.github.io/RPi-Monitor-docs/25_addons.html#).

Like you mentioned, the container itself has no text editor, which leaves you with 2 options

### 1 - add a text editor to the container

```bash
docker exec -it rpi-monitor bash
apt-get update
apt-get install nano
```
### 2 - copy the necessary file(s) to your local machine and overwrite them

Copy original file from container to your current local directory:
```bash
docker cp rpi-monitor:/etc/rpimonitor/data.conf .
```
Copy the modified file to the container and overwrite it:
```bash
docker cp data.conf rpi-monitor:/etc/rpimonitor/data.conf
```

skywinder, 2020-09-04:
Yay. Thank you very much for your help!

closed 2020-09-04 by skywinder