RPi-Monitor/RPi-Monitor #344: Can not get information (dynamic.json) from RPi-Monitor server.

mrblond18, 2020-09-16:
In about 50% of the cases when I open the webpage of the rpi-monitor, I get the proper layout, but no data filled in the fields, with the message:
Can not get information (dynamic.json) from RPi-Monitor server.

It looks like this is from my browser cache. If I refresh I get a 404-page not found.

When I refresh a minute later it works OK with all the data, without restarting the service or doing anything else. It looks like the monitor daeomon or the webpage was "stuck" somehow and cannot serve the data.

How can I investigate further/supply more data...

theWinterDojer, 2020-11-11:
I'm experiencing the same exact issue. Happens about half the time I load the web interface. From what I can tell this only started happening when I upgraded to the Pi 4. I don't recall getting this error on Pi 3.

mrblond18, 2020-11-11:
I am experiencing these problems on a RPi2B...

GlynHu, 2021-01-26:
Thanks for a terrific app in RPI-Monitor Xavier! :)

I thought it was only me seeing this issue!

I am seeing this on all 3 of my Pi's 4B's...sometimes it connects in an instant other times it sits there for up to 60 seconds or so before it connects and then displays exactly as the screenshot in post 1.

Even then a refresh does not always kick it into life.

If there is anything I can do to help track this down please do not hesitate to ask!

Thanks & kind regards,
-=Glyn=-

realestninja, 2024-06-06:
Any updates on this? 

I have two pis, one of them doesn't have this problem, and on the second one, I just installed it moments ago, following the instructions, and got this problem immediately after install, without changing anything on the config.