RPi-Monitor/RPi-Monitor #347: Web server not started because of error: Address already in use

Daronsong, 2020-11-07:
I am still very new to the Raspberry Pi world and I get most of my computer set up done by following others guides, so when I get into situations like this, I have no idea what to do. The error that I wrote in my title is what I got when trying to start rpimonitord and now the service isn't running and got stopped by the system. Any idea what to do to fix it so it runs?

Daronsong, 2020-11-07:
And thank you for your work on this application. When up and running it will be very helpful!

Leepic, 2020-11-28:
Hello,

Maybe because rpimonitord is already running?

Can you give me the output of this command:

`lsof -i :8888`

If rpimonitord or another process is using this port, you will see an output like this one:

`rpimonito 429     1000    4u  IPv4  39461      0t0  TCP *:8888 (LISTEN)`

Here, 429 is the PID of my running rpimonitord process.
You can see the full command of this PID:

```
ps -P 429
  PID PSR TTY      STAT   TIME COMMAND
  429   0 ?        SN     0:00 /usr/bin/perl /usr/bin/rpimonitord
```

Ajimaru, 2021-01-04:
I had the same issue, pigpiod is running at the same tcp port. 

> pi@UnicornPi:~ $ sudo lsof -i :8888
> COMMAND PID USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
> pigpiod 653 root    0u  IPv6  17658      0t0  TCP *:8888 (LISTEN)
> pi@UnicornPi:~` $ ps -P 653
>  PID PSR TTY      STAT   TIME COMMAND
>  653   1 ?        SLsl   0:30 /usr/bin/pigpiod

So I have changed the daemon port of rpi-monitor in /etc/rpimonitor/daemon.conf

> #  daemon.port=8888
> #    Define port of the web server (default:8888)
> daemon.port=8889