RPi-Monitor/RPi-Monitor #348: Needs HTTPS support soon…

Synetech, 2020-11-28:
With browser aggressively moving towards an HTTPS-only web, this is one of the few sites/services that I use which still requires HTTP and now requires special permissions/extra clicks to view. RPi-Monitor will need to run on HTTPS soon.

My server already has an HTTPS certificate which my web-server/blog and Pi-hole use; I don't see why RPi-Monitor couldn't piggyback on that.

wrapper, 2020-11-28:
Surely this is a problem with your configuration rather than RPi Monitor?

Synetech, 2020-11-28:
It's got nothing to do with my configuration. RPi Monitor isn't running under the web service (apache, nginx, etc.), it runs under its own service (using perl), so it needs to separately support HTTPS.

Leepic, 2020-11-28:
As describe in [this URL](https://xavierberger.github.io/RPi-Monitor-docs/34_autentication.html#authentication-and-secure-access), RPi-Monitor will not provide any HTTPS support natively.
You need to setup a web server as a reverse proxy (nginx in the tutorial listed above).

Synetech, 2020-11-29:
It's not ideal, but it's doable. I wasted a lot of time trying to get it to work because I forgot to forward the port in my router. 🤦 Otherwise, it actually takes very little to set it up (even in Apache).

Maybe someday they'll include built-in support as the web evolves and HTTPS becomes mandatory and easier to implement.

psa-jforestier, 2021-04-29:
You mention [this URL](https://xavierberger.github.io/RPi-Monitor-docs/34_autentication.html#authentication-and-secure-access), but it is about adding Nginx webserver in front of RPi-Monitorto provide https and authent.
I did a quick and dirty update directly on RPi-Monitor to add authentication (basic-auth), by modify the `Run` loop of the `Server` package. But I started my modification by directly modifying the code of /usr/bin/rpimonitord (version 2.12-r0), I'm now porting in on the develop branch and do a PR in a few days.