RPi-Monitor/RPi-Monitor #349: RPi-Monitor does not start on Orange Pi Zero2

faand412882, 2020-12-05:
orangepi@orangepizero2:~$ sudo /etc/init.d/rpimonitor status
● rpimonitor.service - LSB: RPi-Monitor daemon
   Loaded: loaded (/etc/init.d/rpimonitor; generated)
   Active: active (exited) since Sun 2020-12-06 01:14:06 MSK; 13min ago
     Docs: man:systemd-sysv-generator(8)
    Tasks: 0 (limit: 1126)
   CGroup: /system.slice/rpimonitor.service

Dec 06 01:14:05 orangepizero2 systemd[1]: Starting LSB: RPi-Monitor daemon...
Dec 06 01:14:05 orangepizero2 rpimonitor[7437]:  * Starting RPi-Monitor
Dec 06 01:14:06 orangepizero2 rpimonitor[7437]: Failed to create share at /usr/bin/rpimonitord line 188.
Dec 06 01:14:06 orangepizero2 rpimonitor[7437]:    ...fail!
Dec 06 01:14:06 orangepizero2 systemd[1]: Started LSB: RPi-Monitor daemon.

orangepi@orangepizero2:~$ sudo lsof -i
COMMAND    PID            USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
systemd-r 2019 systemd-resolve   12u  IPv4   8461      0t0  UDP 127.0.0.53:domain
systemd-r 2019 systemd-resolve   13u  IPv4   8462      0t0  TCP 127.0.0.53:domain (LISTEN)
cupsd     2073            root    6u  IPv6   8890      0t0  TCP localhost:ipp (LISTEN)
cupsd     2073            root    7u  IPv4   8891      0t0  TCP localhost:ipp (LISTEN)
avahi-dae 2076           avahi   12u  IPv4   5809      0t0  UDP *:mdns
avahi-dae 2076           avahi   13u  IPv6   5810      0t0  UDP *:mdns
avahi-dae 2076           avahi   14u  IPv4   5811      0t0  UDP *:49924
avahi-dae 2076           avahi   15u  IPv6   5812      0t0  UDP *:50492
chronyd   2362         _chrony    1u  IPv4   6965      0t0  UDP localhost:323
chronyd   2362         _chrony    2u  IPv6   6966      0t0  UDP localhost:323
sshd      2408            root    3u  IPv4  11046      0t0  TCP *:ssh (LISTEN)
sshd      2408            root    4u  IPv6  11048      0t0  TCP *:ssh (LISTEN)
dhclient  2669            root    6u  IPv4   9589      0t0  UDP *:bootpc
sshd      2830            root    3u  IPv4  11466      0t0  TCP orangepizero2:ssh->192.168.1.54:63674 (ESTABLISHED)
sshd      3274            root    3u  IPv4  13583      0t0  TCP orangepizero2:ssh->192.168.1.54:63754 (ESTABLISHED)
sshd      3417        orangepi    3u  IPv4  13583      0t0  TCP orangepizero2:ssh->192.168.1.54:63754 (ESTABLISHED)

How can I solve this problem?

Ajimaru, 2020-12-17:
Same issue on Raspberry Pi 3+:

pi@octopi:~ $ sudo systemctl status rpimonitor
● rpimonitor.service - LSB: RPi-Monitor daemon
   Loaded: loaded (/etc/init.d/rpimonitor; generated)
   Active: active (exited) since Thu 2020-12-17 11:07:47 CET; 30min ago
     Docs: man:systemd-sysv-generator(8)
    Tasks: 0 (limit: 1601)
   CGroup: /system.slice/rpimonitor.service

Dec 17 11:07:41 octopi systemd[1]: Starting LSB: RPi-Monitor daemon...
Dec 17 11:07:47 octopi rpimonitor[405]: Starting RPi-Monitor:.
Dec 17 11:07:47 octopi systemd[1]: Started LSB: RPi-Monitor daemon.

This happen on every boot, a restart of the service resolve the issue till next boot.

brossow, 2021-01-30:
Similar issue here on a brand-new Pi 3B+ with clean install of Buster using Raspberry Pi Imager directly from Raspberry Pi. It's running alongside SmokePing, which is working as expected and was installed first.

After a reboot, I get this:
```
pi@xxxxx:~ $ /etc/init.d/rpimonitor status
● rpimonitor.service - LSB: RPi-Monitor daemon
   Loaded: loaded (/etc/init.d/rpimonitor; generated)
   Active: inactive (dead)
     Docs: man:systemd-sysv-generator(8)
```

If I try the workaround from @Leepic in #302 of creating a systemd service, I get this instead:
```
pi@xxxxx:~ $ /etc/init.d/rpimonitor status
● rpimonitor.service - RPi-Monitor daemon
   Loaded: loaded (/etc/systemd/system/rpimonitor.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2021-01-29 16:40:00 CST; 5h 30min ago
  Process: 678 ExecStart=/usr/bin/rpimonitord (code=exited, status=3)
 Main PID: 678 (code=exited, status=3)

Jan 29 16:39:59 xxxxx systemd[1]: rpimonitor.service: Main process exited, code=exited, status=3/NOTIMPLEMENTED
Jan 29 16:39:59 xxxxx systemd[1]: rpimonitor.service: Failed with result 'exit-code'.
Jan 29 16:40:00 xxxxx systemd[1]: rpimonitor.service: Service RestartSec=100ms expired, scheduling restart.
Jan 29 16:40:00 xxxxx systemd[1]: rpimonitor.service: Scheduled restart job, restart counter is at 5.
Jan 29 16:40:00 xxxxx systemd[1]: Stopped RPi-Monitor daemon.
Jan 29 16:40:00 xxxxx systemd[1]: rpimonitor.service: Start request repeated too quickly.
Jan 29 16:40:00 xxxxx systemd[1]: rpimonitor.service: Failed with result 'exit-code'.
Jan 29 16:40:00 xxxxx systemd[1]: Failed to start RPi-Monitor daemon.
```

Additional results from various questions/suggestions in that thread:
```
pi@xxxxx:~ $ apt-cache showpkg rpimonitor
Package: rpimonitor
Versions: 
2.12-r0 (/var/lib/apt/lists/giteduberger.fr_rpimonitor_Packages) (/var/lib/dpkg/status)
 Description Language: 
                 File: /var/lib/apt/lists/giteduberger.fr_rpimonitor_Packages
                  MD5: adde0f330c2e7e6ce46cb12f9157aab6
Reverse Depends: 
Dependencies: 
2.12-r0 - perl (0 (null)) librrds-perl (0 (null)) libhttp-daemon-perl (18 6.0.0) libwww-perl (3 6.0.0) libjson-perl (0 (null)) libipc-sharelite-perl (0 (null)) libfile-which-perl (0 (null)) aptitude (0 (null)) 
Provides: 
2.12-r0 - 
Reverse Provides: 

pi@xxxxx:~ $ sudo lsof -i :8888 -n
pi@xxxxx:~ $ 

```
I am not using it behind a reverse proxy, and it works fine if I start it up manually with either `sudo /etc/init.d/rpimonitor start` or `sudo /etc/init.d/rpimonitor restart`. For what it's worth, I have it working perfectly fine on two other 3B+, one running alongside Pi-hole and the other alongside RetroPie, which were originally running Stretch and then upgraded in place to Buster.

 I'm no expert on any of this and don't know what else to try. I'm more than happy to provide any additional info that may be needed. Any advice is appreciated.

peppelongo96, 2021-07-13:
> Similar issue here on a brand-new Pi 3B+ with clean install of Buster using Raspberry Pi Imager directly from Raspberry Pi. It's running alongside SmokePing, which is working as expected and was installed first. I realize this issue was originally opened for an Orange Pi, so I'm happy to split this out into a separate issue if necessary.
> 
> After a reboot, I get this:
> 
> ```
> pi@xxxxx:~ $ /etc/init.d/rpimonitor status
> ● rpimonitor.service - LSB: RPi-Monitor daemon
>    Loaded: loaded (/etc/init.d/rpimonitor; generated)
>    Active: inactive (dead)
>      Docs: man:systemd-sysv-generator(8)
> ```
> 
> If I try the workaround from @Leepic in #302 of creating a systemd service, I get this instead:
> 
> ```
> pi@xxxxx:~ $ /etc/init.d/rpimonitor status
> ● rpimonitor.service - RPi-Monitor daemon
>    Loaded: loaded (/etc/systemd/system/rpimonitor.service; enabled; vendor preset: enabled)
>    Active: failed (Result: exit-code) since Fri 2021-01-29 16:40:00 CST; 5h 30min ago
>   Process: 678 ExecStart=/usr/bin/rpimonitord (code=exited, status=3)
>  Main PID: 678 (code=exited, status=3)
> 
> Jan 29 16:39:59 xxxxx systemd[1]: rpimonitor.service: Main process exited, code=exited, status=3/NOTIMPLEMENTED
> Jan 29 16:39:59 xxxxx systemd[1]: rpimonitor.service: Failed with result 'exit-code'.
> Jan 29 16:40:00 xxxxx systemd[1]: rpimonitor.service: Service RestartSec=100ms expired, scheduling restart.
> Jan 29 16:40:00 xxxxx systemd[1]: rpimonitor.service: Scheduled restart job, restart counter is at 5.
> Jan 29 16:40:00 xxxxx systemd[1]: Stopped RPi-Monitor daemon.
> Jan 29 16:40:00 xxxxx systemd[1]: rpimonitor.service: Start request repeated too quickly.
> Jan 29 16:40:00 xxxxx systemd[1]: rpimonitor.service: Failed with result 'exit-code'.
> Jan 29 16:40:00 xxxxx systemd[1]: Failed to start RPi-Monitor daemon.
> ```
> 
> Additional results from various questions/suggestions in that thread:
> 
> ```
> pi@xxxxx:~ $ apt-cache showpkg rpimonitor
> Package: rpimonitor
> Versions: 
> 2.12-r0 (/var/lib/apt/lists/giteduberger.fr_rpimonitor_Packages) (/var/lib/dpkg/status)
>  Description Language: 
>                  File: /var/lib/apt/lists/giteduberger.fr_rpimonitor_Packages
>                   MD5: adde0f330c2e7e6ce46cb12f9157aab6
> Reverse Depends: 
> Dependencies: 
> 2.12-r0 - perl (0 (null)) librrds-perl (0 (null)) libhttp-daemon-perl (18 6.0.0) libwww-perl (3 6.0.0) libjson-perl (0 (null)) libipc-sharelite-perl (0 (null)) libfile-which-perl (0 (null)) aptitude (0 (null)) 
> Provides: 
> 2.12-r0 - 
> Reverse Provides: 
> 
> pi@xxxxx:~ $ sudo lsof -i :8888 -n
> pi@xxxxx:~ $ 
> ```
> 
> I am not using it behind a reverse proxy, and it works fine if I start it up manually with either `sudo /etc/init.d/rpimonitor start` or `sudo /etc/init.d/rpimonitor restart`. For what it's worth, I have it working perfectly fine on two other 3B+, one running alongside Pi-hole and the other alongside RetroPie, which were originally running Stretch and then upgraded in place to Buster.
> 
> I'm no expert on any of this and don't know what else to try. I'm more than happy to provide any additional info that may be needed. Any advice is appreciated.

Hi man. Are you still facing on it? I have this falling too also after the suggested workaround. I'm on Orange Pi Zero2. Give me the right way if you found it please.

brossow, 2021-07-13:
> Hi man. Are you still facing on it? I have this falling too also after the suggested workaround. I'm on Orange Pi Zero2. Give me the right way if you found it please.

Nope, still having this issue. I simply avoid updates and restart it manually if/as necessary. 😒

peppelongo96, 2021-07-13:
> > Hi man. Are you still facing on it? I have this falling too also after the suggested workaround. I'm on Orange Pi Zero2. Give me the right way if you found it please.
> 
> Nope, still having this issue. I simply avoid updates and restart it manually if/as necessary. 😒

Anyway you are lucky. Not working at all here. I searched for config file and I tried something but all is the same. Howewer, I think that the Leepic fix is right for the rpimonitord run properly. We need some more inspection but I didn't have time :(