RPi-Monitor/RPi-Monitor #350: regular expression not working

a-whatever, 2020-12-17:
Hi,
not sure if it belongs here, but the following regular expression is not working, and I can't figure out why:
dynamic.22.name=typea,typesoa
dynamic.22.source=unbound-control stats_noreset
dynamic.22.regexp=num.query.type.A=(\d+)\nnum.query.type.SOA=(\d+)
dynamic.22.postprocess=
dynamic.22.rrd=

I know I can fix it, by just processing typea and typesoa separately and running unbound-control stats_noreset multible times, but next to these I want to capture more info. So I would like to combine some data request, since it feels unnecessary and unlogical doing multiple calls. 

I tested it on two website, for example https://regex101.com/ where this is perfectly working on the data set below. I just can't figure out why I can't retrieve data from multiple lines in a single call. Is it unsupported? or am I missing something?

mem.cache.rrset=2222
mem.cache.message=2222
mem.mod.iterator=22
mem.mod.validator=222
mem.mod.respip=0
mem.mod.subnet=2222
mem.streamwait=0
histogram.000000.000000.to.000000.000001=2750
histogram.000000.000001.to.000000.000002=0
histogram.000000.000002.to.000000.000004=0
histogram.000000.000004.to.000000.000008=0
histogram.000000.000008.to.000000.000016=0
histogram.000000.000016.to.000000.000032=0
histogram.000000.000032.to.000000.000064=0
histogram.000000.000064.to.000000.000128=0
num.query.type.A=1111
num.query.type.SOA=33
num.query.type.PTR=4444
num.query.type.TXT=20
num.query.type.AAAA=555
num.query.type.SRV=66
num.query.type.TYPE65=666

a-whatever, 2021-01-02:
Hi,

I build a work-around, but still, I hope to get some feedback / response here? 

I have the following solution implemented:
- installed log2ram
- build a script that saves al required data to /var/log/file_xxxx
- crontab job to execute the script every X seconds 
- configured RPI monitor to extract the values from /var/log/file_xxxx
    - file is read many times by RPI monitor, since it contains 15 dynamic values... 

This gives me sometimes duplicated values in the graphs since crontab and RPI are not fully running in sync. but it's okay. At-least all values are matching the same timestamp. 

For me this is still a workaround, so hope that there is a solution for the issue?

albertoacj, 2021-01-14:
I have the same issue here... 
unam -a
Linux nextcloudpi 5.10.5-v8+ #1392 SMP PREEMPT Sat Jan 9 18:56:30 GMT 2021 aarch64 GNU/Linux