RPi-Monitor/RPi-Monitor #353: Installation is not possible anymore : goo.gl  doe not exist  anymore

reba0, 2021-01-12:
Installation   ist no possible anymore : 
 CMD : sudo wget http://goo.gl/vewCLL -O /etc/apt/sources.list.d/rpimonitor.list is not working  suggest  it is as  goo.gl  is not available anymore

sudo wget http://goo.gl/vewCLL -O /etc/apt/sources.list.d/rpimonitor.list
--2021-01-12 19:33:05--  http://goo.gl/vewCLL
Auflösen des Hostnamens goo.gl (goo.gl)… 0.0.0.0, ::
Verbindungsaufbau zu goo.gl (goo.gl)|0.0.0.0|:80 … fehlgeschlagen: Verbindungsaufbau abgelehnt.
Verbindungsaufbau zu goo.gl (goo.gl)|::|:80 … fehlgeschlagen: Verbindungsaufbau abgelehnt.

telemaxx, 2021-02-01:
https://github.com/XavierBerger/RPi-Monitor/issues/354

XavierBerger, 2026-10-27:
Installation process has been redesigned.
See README.md for details.

closed 2026-10-27 by XavierBerger