RPi-Monitor/RPi-Monitor #354: Installation is not possible anymore : http://goo.gl/vewCLL not avail as google disabled goo.gl

reba0, 2021-01-12:
Installation is not possible anymore : 
 CMD : sudo wget http://goo.gl/vewCLL -O /etc/apt/sources.list.d/rpimonitor.list is not working  suggest  it is as  goo.gl  is not available anymore

sudo wget http://goo.gl/vewCLL -O /etc/apt/sources.list.d/rpimonitor.list
--2021-01-12 19:33:05--  http://goo.gl/vewCLL
Auflösen des Hostnamens goo.gl (goo.gl)… 0.0.0.0, ::
Verbindungsaufbau zu goo.gl (goo.gl)|0.0.0.0|:80 … fehlgeschlagen: Verbindungsaufbau abgelehnt.
Verbindungsaufbau zu goo.gl (goo.gl)|::|:80 … fehlgeschlagen: Verbindungsaufbau abgelehnt.

telemaxx, 2021-01-12:
for me the goo.gl link is still working,  but points you to the master branch. the rpimonitor.list is not working anymore. the developerbranch is ok.
https://github.com/XavierBerger/RPi-Monitor/tree/develop/src/etc/apt/sources.list.d

telemaxx, 2021-01-12:
maybe the author could merge this file into the master. than the goo.gl links are ok again.

XavierBerger, 2026-10-27:
Installation process has been redesigned.
See README.md for details.

closed 2026-10-27 by XavierBerger