RPi-Monitor/RPi-Monitor #356: get storage info by-uuid

Darkyere, 2021-01-17:
I Am trying to configure storage.conf to show info on two partitions.
Now it works fine if I just use sda1 etc.

But I have this little problem that I sometimes have multiple HDD's connected to a HUB on a RPI4 and rebooting my pi it changes the order I of /dev/sdX

meant like what was before /dev/sda1 might be /dev/sdc1 on next reboot.

I was hoping to use /dev/disk/by-uuid/XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX

But I can't figure out how to use that in storage.conf

Any ideas of how to do that ?

Feriman22, 2021-03-05:
Hi,

In `/dev/disk/by-uuid` there are symlinks mapping each drive's UUID to its entry in /dev (e.g. `/dev/sda1`)

So, first run this command to see your uuid:

`ls -la /dev/disk/by-uuid`

Then try it like this way:

`df -h /dev/disk/by-uuid/9b39e298-250e-4688-b41c-6dc518c954b9`

Darkyere, 2021-03-05:
I dont quite understand how to make use of this info.
If i look at 

```
static.10.name=storage1_total
static.10.source=df -t ext4
static.10.regexp=sda9\s+(\d+)
static.10.postprocess=$1/1024
```

# And changes this to

```
static.10.name=storage1_total
static.10.source=df -h d864bdfe-fe9a-4c68-8589-bdccad9cbbb1
static.10.regexp=sda9\s+(\d+)
static.10.postprocess=$1/1024
```

I get

```
Syncthing Used: 52.14GB (NaN%) Free: NaNkB Total: NaNkB

NaN%
```

It seems  like i get the total available space from df -h
But what do i do about

`static.10.regexp=sda9\s+(\d+)`

----

This was origionally the command part i had problem chaning it doesnt seems to accept UUID in the ways i tried so far.

Thank you for writing back i hope u got a solution that i aint able to see.

Best Regards,
Darkyere

Feriman22, 2021-03-05:
Change this line from:

`static.10.source=df -h d864bdfe-fe9a-4c68-8589-bdccad9cbbb1`

to

`static.10.source=df -h /dev/disk/by-uuid/d864bdfe-fe9a-4c68-8589-bdccad9cbbb1`

Darkyere, 2021-03-05:
Sort of looks like its still not quite there yet

`Syncthing Used: 52.14GB (22687415.77%) Free: -52.14GB Total: 241.00kB`

And the command

`static.10.regexp=sda9\s+(\d+)`

stiil points to sda9 so if the disk order changes i think there will be some sort of missmatch

Feriman22, 2021-03-05:
I'm using it in this way:

```
static.1.name=storage1_total
static.1.source=df -t ext4
static.1.regexp=sda1\s+(\d+)
static.1.postprocess=$1/1048576
```

So you should to use it like this:

```
static.10.name=storage1_total
static.10.source=df /dev/disk/by-uuid/d864bdfe-fe9a-4c68-8589-bdccad9cbbb1
static.10.regexp=sda9\s+(\d+)
static.10.postprocess=$1/1048576
```

Darkyere, 2021-03-05:
Okay this is getting somewhere now im getting 

```
Syncthing Used: 52.14GB (21.71%) Free: 188.04GB Total: 240.18GB

21.71%
```

----

But what i worry about is when the pi reboots and the order of /dev/sdXy changes.

At the moment the command

`df /dev/disk/by-uuid/d864bdfe-fe9a-4c68-8589-bdccad9cbbb1`

reports back 

```
Filesystem     1K-blocks     Used Available Use% Mounted on
/dev/sda9      251850484 54676672 184357428  23% /media/cfna08h/Syncthing
```

the regex command does

`static.10.regexp=sda9\s+(\d+)`

so when i reboot and /dev/disk/by-uuid/d864bdfe-fe9a-4c68-8589-bdccad9cbbb1
is no longer sda9 but sdc9 then the command will fail.

I allso needs this part of the command to match up some how.

Like a regex expression that just reads 3 and 4 line in the second collum since its the only reported device by df -h.

So that it is persistent even if /dev/sdXy path changes.

Feriman22, 2021-03-05:
Simply modify the regex from

`static.10.regexp=sda9\s+(\d+)`

to

`static.10.regexp=dev\s+(\d+)`

Darkyere, 2021-03-05:
Sry it just throws me back to

```
Syncthing Used: 52.14GB (NaN%) Free: NaNkB Total: NaNkB

NaN%
```

Feriman22, 2021-03-05:
Yeah, true. Try this one:

`static.10.regexp=sd\s+(\d+)`

Darkyere, 2021-03-05:
It seems to give the same result.

```
Syncthing Used: 52.14GB (NaN%) Free: NaNkB Total: NaNkB

NaN%
```