RPi-Monitor/RPi-Monitor #357: The repository 'https://giteduberger.fr rpimonitor/ Release' no longer has a Release file.

MatthK, 2021-02-01:
I just installed RPi-Monitor using the getting started guide and now have the problem that my `sudo apt update` doesn't work anymore.

I do get the following error message:

```
pi@Pihole:~ $ sudo apt update
Hit:1 http://packages.microsoft.com/repos/code stable InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Hit:3 http://raspbian.raspberrypi.org/raspbian buster InRelease
Ign:4 https://giteduberger.fr rpimonitor/ InRelease
Err:5 https://giteduberger.fr rpimonitor/ Release
  Redirection loop encountered
Reading package lists... Done
E: The repository 'https://giteduberger.fr rpimonitor/ Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

How can I get the update working again?

carlosala, 2021-02-01:
Same issue

telemaxx, 2021-02-01:
https://github.com/XavierBerger/RPi-Monitor/issues/354

MatthK, 2021-02-02:
It still does not work. This is my /etc/apt/sources.list.d/rpimonitor.list file:

```
# RPi-Monitor official repository
deb https://giteduberger.fr rpimonitor/
```
And with that, I do get the error message.

pkloehn41, 2021-02-02:
Putting the trailing '/' behind the domain allowed the install to complete for me.

`# RPi-Monitor official repository
deb https://giteduberger.fr/ rpimonitor/`

MatthK, 2021-02-02:
That indeed did the trick. Thanks.

closed 2021-02-02 by MatthK

ardinusawan, 2022-10-17:
This did the trick https://forums.raspberrypi.com/viewtopic.php?t=255932#p1823865

```
# RPi-Monitor official repository
deb https://www.giteduberger.fr rpimonitor/
```