RPi-Monitor/RPi-Monitor #359: CPU Temperature range: How to start from 0C rather than 40C

spxak1, 2021-02-18:
Currently the CPU temperature range is 40-80C and I can see why, however at lower temperatures the gauge doesn't show anything visual.

What is a way to change the range of that gauge to go from 0C to 80C? I appreciate that 0C is not realistic, but it will offer a better scale to that gauge, that will put 40ish C right in the middle, with lower temperatures than 40C visually shown as less than half, and more than 40C more than half. An easy way to tell what's happening with one look.

Thank you.

telemaxx, 2021-02-18:
edit this file:
/etc/rpimonitor/template/temperature.conf

find this line:
web.status.1.content.1.line.1=JustGageBar("CPU Temperature", data.soc_temp+"°C", 40, data.soc_temp, 80, 100, 80)

to:
web.status.1.content.1.line.1=JustGageBar("CPU Temperature", data.soc_temp+"°C", 0, data.soc_temp, 100, 100, 80)

https://xavierberger.github.io/RPi-Monitor-docs/24_web.html

telemaxx, 2021-02-18:
don't forget to restart service or complete PI

spxak1, 2021-02-18:
> edit this file:
> /etc/rpimonitor/template/temperature.conf
>
> find this line:
> web.status.1.content.1.line.1=JustGageBar("CPU Temperature", data.soc_temp+"°C", 40, data.soc_temp, 80, 100, 80)
>
> to: [...]

Worked like a dream. Many thanks!!!

telemaxx, 2021-02-18:
You're welcome