RPi-Monitor/RPi-Monitor #367: Statistics graph with wrong datetime

monojk, 2021-05-18:
I have rpimonitor running on 3 Raspberries. On all, the datetime on the x-axis is very different but never recent, even after reset selection was pressed. Today is May 18th
The rpimonitors were all installed and activated on the same day.  Info for the 2 Raspberries:
`Processor: BCM2835
Distribution: Raspbian GNU/Linux 10 (buster)
Kernel version: Linux 5.10.17-v7+ armv7l
Firmware: #1403
Package(s): 15 upgradable(s)`

`Processor: BCM2835
Distribution: Raspbian GNU/Linux 10 (buster)
Kernel version: Linux 5.10.17-v7+ armv7l
Firmware: #1403
Package(s): 16 upgradable(s) `
 
[image]

telemaxx, 2021-05-18:
have you checked Date/Time eg via console?

monojk, 2021-05-18:
In the status tab of rpimonitor as well as in the console the date and time is correct.

telemaxx, 2021-05-19:
strange.

psa-jforestier, 2021-05-26:
Hello. Can you check what is the date in the /var/log/rpimonitor.log file ?
Also, add some verbosity to the log by doing `sudo killall -USR1 rpimonitor` and check the date displayed on the log file.

monojk, 2021-05-26:
pi@rp3b0:/var/log $ ls -latr rpimonitor.log
    -rw-r--r-- 1 root root 224 May 24 12:33 rpimonitor.log

    pi@rp3b0:/var/log $ sudo killall -USR1 rpimonitord

    pi@rp3b0:/var/log $ ls -latr rpimonitor.log
    -rw-r--r-- 1 root root 353 May 26 09:52 rpimonitor.log

    pi@rp3b0:/var/log $ tail -f rpimonitor.log
    [2021/05/24-12:33:45] RPi-Monitor has been started
                      Log level can be increased or decreased during runtime
                      by sending signal USR1 or USR2 with the command killall -USRx rpimonitord
    [2021/05/26-09:52:51] Change loglevel to 1
    [[2021/05/26-09:52:512021/05/26-09:52:51] Change loglevel to ] Change loglevel to 11

monojk, 2021-06-05:
Any ideas?

monojk, 2021-06-07:
I see that when selecting "Uptime", "CPU Loads", or "Temperature", the charts have the correct date/time in contrast to "Memory", "Swap", and "Disks"

telemaxx, 2021-06-07:
can try to stop rpimonitor, rename one of the rrd database files and start the service again. /var/lib/rpimonitor/stat/swap.rrd

monojk, 2021-06-07:
stopped rpimonitor
renamed sdcard_root_used.rrd 
started rpimonitor
sdcard_root_used.rrd gets created but the graph is the same as before for "Disks - root" 
all rrd files have the same size

gilarelli, 2021-06-09:
Try to put the "Total Memory" graph (the _static_ one) **not** in first place : 

> web.statistics.1.content.1.name=Memory
> web.statistics.1.content.1.graph.1=memory_available
> **web.statistics.1.content.1.graph.2=memory_total**
> web.statistics.1.content.1.ds_graph_options.memory_available.label=Available Memory (GB)
> **web.statistics.1.content.1.ds_graph_options.memory_total.label=Total Memory(GB)**

I had the same troubles as you and found this solution on the web (can't remember where tho)

monojk, 2021-06-10:
> Try to put the "Total Memory" graph (the _static_ one) **not** in first place :
>
>
> I had the same troubles as you and found this solution on the web (can't remember where tho)

Which file(s) would I have to change? I was looking into the files under /etc/rpimonitor and /etc/rpimonitor/template

gilarelli, 2021-06-10:
> Which file(s) would I have to change? I was looking into the files under /etc/rpimonitor and /etc/rpimonitor/template

In the .conf file for the metrics you want to graph that is called in your /etc/rpimonitor/data.conf with an "include" line

monojk, 2021-06-11:
Thanks, that did the trick.

gilarelli, 2021-06-11:
> Thanks, that did the trick.

You're welcome :)

>  So, these statistics are affected: Memory, Swap, Disks - boot, Disks - root

Basically, each time you want to have a static value in your graph, you have to put it **NOT** in first place

monojk, 2021-06-11:
> > Thanks, that did the trick.
> 
> You're welcome :)
> 
> > So, these statistics are affected: Memory, Swap, Disks - boot, Disks - root
> 
> Basically, each time you want to have a static value in your graph, you have to put it **NOT** in first place

So, yes. Initially I didn't change anything from the fresh install and got the problem. I guess it needs to be fixed there also.