RPi-Monitor/RPi-Monitor #372: Is RPIMonitor development at a halt?

wsegatto, 2021-06-28:
Hey everyone, 

I don't want to be rude or anything, but I'm wondering whether RPIMonitor development is still ongoing or if it has been stopped.
It seems there are 5+ years old requests unanswered and I see no new versions coming out...
I use it every day and it's fundamental for my monitoring purposes, but I wanted to have your input.

This is my setup:
[image]

4k3or3et, 2021-11-13:
I would like to know that also.

tegeek1445, 2022-02-01:
> Hey everyone,
>
> I don't want to be rude or anything, but I'm wondering whether RPIMonitor development is still ongoing or if it has been stopped. It seems there are 5+ years old requests unanswered [...]

Hey WIll can I know how do you configure rpi monitor to show live network usage.
(i'm running rpi monitor in armbian on asus tinker board using usb tethering from my old phone and the sbc as my router if that matters)

TIA

wsegatto, 2022-02-01:
Hi @tegeek1445, I'm using native eth0 statistics of my linux box. For the other parameters I'm running bash or python scripts to write out to a TXT not to overload RPI with too many processes and calculations.

```
dynamic.10.name=net_received
dynamic.10.source=/sys/class/net/eth0/statistics/rx_bytes
dynamic.10.regexp=(.*)
dynamic.10.postprocess=$1*-1
dynamic.10.rrd=DERIVE
dynamic.10.max=0
 
dynamic.11.name=net_send
dynamic.11.source=/sys/class/net/eth0/statistics/tx_bytes
dynamic.11.regexp=(.*)
dynamic.11.postprocess=
dynamic.11.rrd=DERIVE
dynamic.11.min=0
```

Full network file here:  https://pastebin.com/6erjuzeW

wsegatto, 2022-02-01:
Oh, and the panel now looks like this:
[image]

asyba, 2023-01-15:
> Hey everyone,
>
> I don't want to be rude or anything, but I'm wondering whether RPIMonitor development is still ongoing or if it has been stopped. It seems there are 5+ years old requests unanswered [...]

niceee,
do you know if possible to show data from doing the command top?
that show top process that consume ram, cpu, etc to monitor that?

XavierBerger, 2026-10-27:
**RPi-Monitor** development was in pause and has officially restart. See README.md for details