RPi-Monitor/RPi-Monitor #374: CPU Temperature not working with 2.13-beta6

SabineTobi, 2021-08-03:
Hi!
With version 2.13-beta6 I can not monitor the CPU Temperature with the template temerature.conf. RPi-Monitor is showing only NULL as temperature!
With version 2.12-r0 it does as expected.

Regards,
Sabine

Lusbueb69, 2021-08-19:
Same Problem and my **_Workaround_** is this, with all digits:

Change Line 14 in the **temperature.conf** from:

`dynamic.1.postprocess=sprintf("%.2f", $1/1000)`

to

`dynamic.1.postprocess=$1/1000`

hackslikeus, 2021-11-27:
for those of us still using ye ol fair in height:
dynamic.1.name=soc_temp
dynamic.1.source=/sys/devices/virtual/thermal/thermal_zone0/temp
#dynamic.1.sourc=cat /sys/class/thermal/thermal_zone0/temp
dynamic.1.regexp=(.*)
dynamic.1.postprocess=($1*.001)
#dynamic.1.postprocess=sprintf("%.2f", $1/1024)
dynamic.1.rrd=GAUGE

dynamic.20.name=sof_temp
dynamic.20.source=/sys/devices/virtual/thermal/thermal_zone0/temp
dynamic.20.regexp=(.*)
dynamic.20.postprocess=((($1/1000)*1.8)+32)
dynamic.20.rrd=GAUGE

beohna, 2021-12-08:
Same problem with the CPU Temperature (2.13-beta6 ).

Why is there no new beta?

beohna, 2021-12-08:
The workarounds above didn't work on my raspi 4B.

JappeHallunken, 2022-02-02:
Another workaround, shows no decimal places at all:
`dynamic.1.postprocess=floor($1/1000)`

JappeHallunken, 2022-02-02:
Or maybe ceil() would be the better choice for temperatures, because it's rounds the value up.

MichaIng, 2022-02-02:
The problem is the "new" ability to define multiple postprocess steps comma-separated: https://github.com/XavierBerger/RPi-Monitor/commit/56ba6e9#diff-ba4b03131626dc7d144ab66856ad8f9306bad3c958e5a20cda2a0a9e655208d4R777

Since `sprintf("%.2f", $1/1000)` contains a comma, it is broken and evaluated in two parts 😄. So yeah, we need a function/method which does not contain any comma. Instead of `lower()`, I suggest `int($1/1000 + 0.5)` to have correct rounding instead of always rounding down. But better would be something with decimal support, of course. I'm no perl coder, but will have a look into possibilities.

wsegatto, 2022-08-01:
Better yet, @MichaIng, if you want decimal places with the 'int' function:

int($1/1000) --> No decimal places
int($1/100)/10 --> 1 decimal place
int($1/10)/100 --> 2 decimal places

Why do you mean when you said "'new' ability"? What's changed? RPIMonitor? There hasn't been a commit for so long now. I wonder what you meant by that. Just curious.

MichaIng, 2022-08-01:
@wsegatto 
Thanks for your suggestion. But this will always round down, isn't it? The `+ 0.5` assures it is mathematically correctly rounded, so up from `.5` on.

> Why do you mean when you said "'new' ability"?

Yeah, the commit is from 2018, hence the quotation marks 😉. Interesting that the bug was reported 3 years later the first time 🤔.

MichaIng, 2022-08-01:
Ah lol, we went that way already. Too long ago so I forgot about the details: https://github.com/MichaIng/DietPi/pull/5232/files

XavierBerger, 2026-10-31:
Fixed in v2.13.1

closed 2026-10-31 by XavierBerger