RPi-Monitor/RPi-Monitor #385: RPi-Monitor on Raspbian OS Bullseye.

4k3or3et, 2021-11-13:
Hi All

i just installed RPi-Monitor on Raspbian OS Bullseye. Exactly the same config like on Buster. The problem i am facing on new OS are graphs. For example uptime was always linear. Now it behaves like zig-zag. Very often next value of uptime is represented as lower that previous which does not make sense. 

[image]
[image]

The same is for other values like SOC or RAM voltages where the values all 99% of time at exactly same level.

Anyone have a clue how to fix that?

gworley, 2021-11-23:
Hi this doesn't answer your question.  I cannot get it to install on my Raspberry Pi  running bullseye.

4k3or3et, 2021-12-03:
Anyone?

sborn, 2022-03-30:
I have it running and behaving normally on Bullseye on 16 pis. I don't imagine that helps you guys, but maybe the bugs have been resolved?

adi170-alt, 2022-04-04:
> Hi this doesn't answer your question. I cannot get it to install on my Raspberry Pi running bullseye.

oof, it works on mine, but my pi crashed a few times, but that was because I was trying to overclock my pi

gilarelli, 2022-05-13:
Hi,

Got the same trouble with most of my graphs - also after upgrading from buster to bullseye... 

As an example, this is a graph that should show the number of IPs banned by fail2ban - it's a simple iptables interrogation that should always return an integer:
```
dynamic.6.name=banned
dynamic.6.source=iptables -L -n | awk '($1=="DROP" || $1=="REJECT") && $4!="0.0.0.0/0" {print $4}' | sort | uniq | wc -l
dynamic.6.rrd=GAUGE
```

Here is the result on the status page:
[image]

And here on the statistic page:
[image]

Detail of the values:
[image]

Here is another example, the graph for the root fs use:
[image]