RPi-Monitor/RPi-Monitor #391: Can't update RPi Monitor

Machou, 2022-03-01:
Hello

When i want update RPi Monitor on Raspbian, i have this error :

Err :6 http://giteduberger.fr rpimonitor/ InRelease
  Connexion à giteduberger.fr: 80 (213.186.33.17) impossible. - connect (113: Aucun chemin d'accès pour atteindre l'hôte cible)

marapavelka, 2022-03-03:
Edit /etc/apt/sources.list.d/rpimonitor.list and change line `deb http://giteduberger.fr rpimonitor/` to `deb https://www.giteduberger.fr rpimonitor/`

Machou, 2022-03-09:
I have this error now :

Err :10 https://www.giteduberger.fr rpimonitor/ InRelease
  Connexion à www.giteduberger.fr: 443 (213.186.33.17) impossible. - connect (113: Aucun chemin d'accès pour atteindre l'hôte cible)

`cat /etc/apt/sources.list.d/rpimonitor.list`

```bash
# RPi-Monitor official repository
deb https://www.giteduberger.fr rpimonitor/
```

closed 2022-03-14 by Machou