RPi-Monitor/RPi-Monitor #393: Perl error on Raspberry pi bullseye 64-bit

racedowling, 2022-03-27:
I get this log error each time rpimonitor runs:
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN1405 at /usr/share/perl5/HTTP/Daemon.pm line 344.

I have reinstalled rpi 2.13 and all the perl modules. Rpimonitor is working. I don't know if this is an rpimonitor or a perl issue. 
Any clues would be appreciated.

mushu999, 2024-01-18:
Also, see this issue which shows it works in Bullseye: https://github.com/XavierBerger/RPi-Monitor/issues/385

Vyacheslav-S, 2024-04-10:
I've solved this problem - I've changed **_rpimonitord.service_**:
```
[Unit]
Description=RPi-Monitor daemon
Before=multi-user.target
After=remote-fs.target
After=network-online.target
Wants=network-online.target

[Service]
Type=simple
Restart=on-failure
KillMode=mixed
Nice=19
ExecStart=/usr/bin/rpimonitord
ExecStop=/bin/kill $MAINPID
StandardOutput=append:/var/log/rpimonitor.log
StandardError=append:/var/log/rpimonitor.log

[Install]
WantedBy=multi-user.target

```