RPi-Monitor/RPi-Monitor #401: Service Status via Labels and Bages, not being updated when service is brough down

thygizmo, 2022-09-05:
Hi,
Ever since some years ago, I have noticed that the "service monitoring" implementation, actually shows the badges/labels as "Red"/down if the ports for instance are actually closed.

Starting up a service (e.g.: FTP service on port 21), actually makes the label/badge change from **red** to **green** status, which is ok and expected. However, if I stop an FTP server listening on port 21 (e.g.: netstat correctly reports there is no port being LISTENed), but the Label/Badge does not change back to **red**...

Does anyone know if this is a known issue, or I am doing something wrong, somewhere?

Here's the relevant configuration that I have setup for the ftp Service monitoring:

dynamic.1.name=ftp
dynamic.1.source=netstat -nlt
dynamic.1.regexp=tcp .*\:*(21).*LISTEN

web.status.1.content.1.line.1=Badge(data.ftp,"==21","ftp(21)","success") + Badge(data.ftp,"!=21","ftp(21)","danger") 

Many thanks for any tips you might have...:)

Cheers,
Gizmo

JappeHallunken, 2022-09-17:
I've got this working with labels, and I use `ss`, since `netstat`is deprecated.
Your `web.status` stuff looks right, so I guess the error happens maybe at the regexp?
Here my code for comparison:

```
dynamic.3.name=http
dynamic.3.source=ss -nltp
dynamic.3.regexp=LISTEN .+:(80).+
dynamic.3.default=0

web.status.1.content.1.line.2="<b>Lighttpd HTTP</b>: "+Label(data.http,"==80","listening","success")+Label(data.http,"!=80","not listening","danger")
```

thygizmo, 2022-09-18:
Hi! Thanks for the feedback and for your piece of code!
Actually the regexp is **correct** and matches the output of the netstat command!
However, I did not have in my configuration the "dynmic.x.default=0 value" and that made the whole difference!!!

Thank you for your feedback, I finally got it fully working!

Many thanks!
Giz

JappeHallunken, 2022-09-19:
Nice that you found a solution.
It's a bit offtopic, but I'm still wondering about the regexp.
Because when I test this [https://regexr.com/6ub3v](url), it doesn't match without the + and I don't understand why.