RPi-Monitor/RPi-Monitor #406: Perl issue appear in the log

Feriman22, 2023-01-11:
I use the latest RPi-Monitor (v2.12), and I have some errors in the log. It seems the service works well, I didn't notice any issue with that.

Debian 11.6 used on my Raspberry Pi 4.

Full log:

```
[2023/01/10-14:20:09] RPi-Monitor has been started
                      Log level can be increased or decreased during runtime
                      by sending signal USR1 or USR2 with the command killall -USRx rpimonitord
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN1 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN2 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN3 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN4 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN5 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN6 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN7 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN8 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN9 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN10 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN11 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN12 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN13 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN14 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN15 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN16 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN17 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN18 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN19 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN20 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN21 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN22 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN23 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN24 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN25 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN26 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN27 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN28 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN29 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN30 at /usr/share/perl5/HTTP/Daemon.pm line 372.
[... 236 lines cut ...]
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN149 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN150 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN151 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN152 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN153 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN154 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN155 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN156 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN157 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN158 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN159 at /usr/share/perl5/HTTP/Daemon.pm line 372.
Use of uninitialized value in vec at /usr/share/perl5/HTTP/Daemon.pm line 111.
sysread() on closed filehandle GEN160 at /usr/share/perl5/HTTP/Daemon.pm line 372.
```

fotosik, 2024-02-15:
Hi,

I use Version: 2.13-beta6 by Xavier Berger,
I have the same issue as above

Maybe someone has a solution to the phenomenon...

jaapeldoorn, 2024-05-17:
This event is triggered by the line $this->SendJSON($var); (line number approx 436) in /usr/bin/rpimonitord .
One example of the $var is:
2[2024/05/17-20:00:52]       Server::SendStatus JSON JV1: ```
{"rpimonitor":"8888","uptime":"355541.37","scaling_governor":"ondemand","soc_temp":"53.69","Recidive_Bans":"0","ssh":"22","wifi_received":-1063072490,"Total_Jails":"21",sdcard_used":63.08984375,"storage1_used":59997.98828125,"storage2_used":6132.05078125,"http":"80","smtp":"25","smtps":"587","localtime":[2024,5,17,20,0,46],"https":"443","memory_available":503.40234375,"cpu_voltage":"1.2875",
"mysql":"3306","load1":"0.30","sieve":"4190","postfix_sasl_Bans":"0","cups":"631","memory_free":185.640625,"redis":"6379","wifi_send":"472457109","sdcard_total":509.990234375,"upgrade":"0 upgradable(s)","load5":"0.33",
"intprint":"631","storage3_used":677441.25390625,"samba":"445","load15":"0.27","packages":"","swap_used":183.90234375,"cpu_frequency":1200,"imaps":"993","sshd_Bans":"0","imap":"143","smb":"139"}