RPi-Monitor/RPi-Monitor #407: unable to update RPi-Monitor to version 2.13

Schnappa, 2023-01-24:
I followed the instructions to update on [https://xavierberger.github.io/RPi-Monitor-docs/11_installation.html] but the line `sudo /etc/init.d/rpimonitor update` shows only this after a while in the terminal:

`[ ok ] RPi-Monitor update packages status:.`

The version remains on 2.12.

losgobbi, 2023-02-10:
Hi, the 2.12-r0 is the latest version as we can see [here](https://github.com/XavierBerger/RPi-Monitor-deb/tree/master/packages) There were no more new releases after the 2.12 version. I think Xavier is not maintaining this project anymore.

mushu999, 2024-01-18:
Sorry to necro this thread, but for indexing purposes I believe the final version is 2.13 since I'm running: **Version: 2.13-beta6 by Xavier Berger**

Itsisme, 2024-07-07:
Hai,

Does anyone have a link to the 2.13-beta6 version ?

Rgds...

losgobbi, 2024-07-08:
It could be this debian pkg: https://github.com/XavierBerger/RPi-Monitor-deb/blob/87cf9d2af9c6a521cc17d05f757a46693026281f/repo/rpimonitor_2.13-beta6_all.deb
Tks!!!